bit-bots/wolfgang_robot #77: Simplified collision models instead of Bounding Boxes in webots model

jgueldenstein, 2021-01-19:
Currently we use axis aligned bounding boxes as a collision model.

While this works in most cases, it prevents the standup from working correctly. 

Using a simplified version of the visual model might be preferable. Convex hull is a good starting point. after that maybe mesh simplification. Meshlab has these functions but it would be nice to have a script perform the operations for each .stl file automatically. The URDF also has to be adjusted.

We can integrate this either in onshape2robot or as a separate script.

SammyRamone, 2021-01-20:
It is possible to further reduce the size of the stls, see https://onshape-to-robot.readthedocs.io/en/latest/config.html
maybe this would already give a good simplification of the model

jgueldenstein, 2021-01-21:
This is still a manual solution and requires a lot of work for each component.

closed 2021-02-17 by jgueldenstein