bit-bots/wolfgang_robot #81: Linear acceleration computation in pybullet sim and gimbal lock

SammyRamone, 2021-02-05:
The linear acceleration provided by the pybullet sim does not work, due to gimbal lock issues. The gravity vector is added based on euler angles and this results in wrong values if the robot is lying on the floor.

## Expected behavior
Correct acceleration values which are not higher than 9.81 if robot is not moving.

## Current behavior
acceleration > 30

## Steps to Reproduce
1. Put robot on its belly in simulation
2. observe imu values in plotjuggler

## Context (Environment)

- [ ] RViz
- [ ] Simulator
- [ ] Robot
- [ ] Local

## Possible Solution

SammyRamone, 2021-04-26:
currently gravitation is computed as
```
       r,p,y = euler_from_quaternion(orientation)
        gravity = [r*9.81,p*9.81,y*9.81]
```
which is obviously wrong since each euler angle is bound between [-tau/2, tau/2]

jaagut, 2022-11-17:
Is this now completely fixed?

closed 2022-11-17 by jaagut