bit-bots/wolfgang_robot #101: Initial Poses of robot model

jgueldenstein, 2021-03-03:
We could have the model load in, already being in the walkready pose. This seems to be relatively easy achieved with a command line argument to the urdf2webots script. What do you think about this? @timonegk @SammyRamone

timonegk, 2021-03-04:
I would like that because the current init pose is just the random pose where all joint positions are zero and which happens to be stable for the robot. We don't use that pose on the real robot therefore I see no need in having it in the simulation, especially when it can be changed as easily as you say.

SammyRamone, 2021-03-04:
Yeah me too

SammyRamone, 2021-03-24:
doing this via urdf2webots would maybe prevent us from using the spring in the knee. this must be evaluated. https://github.com/bit-bots/wolfgang_robot/issues/116

jgueldenstein, 2021-03-25:
as said by @SammyRamone , this probably messes with the spring and it does not have much benefit otherwise other than robots looking nice when no controller is running.

closed 2021-03-25 by jgueldenstein