bit-bots/wolfgang_robot #118: evaluate hinge-2 to speed up webots

SammyRamone, 2021-03-23:
It looks like webots uses hinge-2 joints to speed up simulation of darwin. We could do the same
https://hl.forum.robocup.org/t/first-webots-package-for-the-virtual-hsc/481/7

SammyRamone, 2021-03-24:
see also https://github.com/cyberbotics/webots/pull/2861

SammyRamone, 2021-03-24:
looks like this could actually increase the performance by around 37%. Unfortunatly, it is not easy to do, since we can not model a 2axis rotation joint in the URDF. We therefore need to do this later in the URDF2Webots or after that. Maybe using a simple "find and replace" script on the final proto is the simplest approach

jgueldenstein, 2021-03-25:
Since the Hinge2Joint would not allow us to rotate the motors independently, this would look quite wired and be really hacky.

Also it is a pain to implement since all transforms would have to be recomputed.

closed 2021-03-25 by jgueldenstein