bit-bots/wolfgang_robot #184: our Wolfgang RoboCup Api can't handle ipv6 adresses

johagge, 2021-06-09:
Our node for the RoboCup Api crashes when using an IPv6 adress as simulator IP.

## Expected behavior

We should probably use everything until the last ':' for host and only the last one as port.
Alternatively I would be fine if this is closed as won't fix.

## Current behavior

File "/home/jonas/git/bitbots/bitbots_meta/rosws/src/bitbots_meta/wolfgang_robot/wolfgang_robocup_api/src/wolfgang_robocup_api/wolfgang_robocup_api.py", line 136, in get_connection
    host, port = addr.split(':')
ValueError: too many values to unpack (expected 2)

## Steps to Reproduce

1. set ROBOCUP_SIMULATOR_ADDR=some:ipv6:addr
2. rosrun wolfgang_robocup_api start.sh goalie 0
3. look at the error
...

## Context (Environment)

- [ ] RViz
- [X] Simulator
- [ ] Robot
- [ ] Local

## Possible Solution

Flova, 2021-06-09:
I think this is a do not fix

SammyRamone, 2022-07-01:
@timonegk what do you think? do we need to fix this or do we close this issue?

SammyRamone, 2022-07-01:
maybe also influenced by https://github.com/bit-bots/wolfgang_robot/issues/217

timonegk, 2022-07-01:
If we want to continue using our api, we should fix it. The fix is very simple, just replace `split(":") ` by rsplit(":", 1)`.