bit-bots/wolfgang_robot #250: Remove IK solvers

SammyRamone, 2022-12-20:
We changed the kinematic config here to get around a bug
https://github.com/bit-bots/wolfgang_robot/pull/248
It would make more sense to have no solvers for this groups but this was just merged and is not in the packages yet
https://github.com/ros-planning/moveit2/pull/1816

## Expected behavior

## Current behavior

## Steps to Reproduce

1.
2.
3.
...

## Context (Environment)

- [ ] RViz
- [ ] Simulator
- [ ] Robot
- [ ] Local

## Possible Solution

jaagut, 2023-08-09:
Is this still relevant?

SammyRamone, 2023-08-17:
> Is this still relevant?

Yes. Currently you load the BioIK solver for each move group which takes a lot of time during initialization. Some of these groups are not used to provide IK solutions but just to get joint values of certain robot parts. It would be better to just load the default KDL solver which is quicker to load.

timonegk, 2023-08-17:
The fix in moveit you mentioned in the issue is released in version 2.7.0, so we can probably revert the other PR.

Flova, 2023-08-30:
> Yes. Currently you load the BioIK solver for each move group which takes a lot of time during initialization. Some of these groups are not used to provide IK solutions but just to get joint values [...]

This might explain the extremly long motion startup times we had on the WorldCup!

PR #281 (2023-08-30): Revert "Use bio_ik instead of the KDL default for all move groups"
This reverts commit c061f0519a9a038191ef92071313f2cd19ea98c1.

Closes #250
closed 2023-08-30 by Flova