bit-bots/wolfgang_robot #268: ROS-bag record dies, when output folder already exists

jaagut, 2023-03-26:
The recording of an ROS-bag dies even in the HLVS setup because the output folder already exists. This may have happened, as the game was started multiple times.

This may also apply to bit-bots/bitbots_misc/pull/190

## Expected behavior
The recording should not stop and die.

## Current behavior
```
[ros2-2] [ERROR] [ros2bag]: Output folder '/robocup-logs/ID_1' already exists.
[hlvs_player-3] Failed to connect to 172.31.6.199:10021 (attempt 3)
[ERROR] [ros2-2]: process has died [pid 63, exit code 1, cmd 'ros2 bag record -o /robocup-logs/ID_1 --include-hidden-topics /ball_position_relative_filtered /ball_obstacle_active /ball_relative_filtered /ball_relative_movement /balls_in_image /balls_relative /bitbots_localization/pose_particles /camera/images_to_log /camera_info /clock /cmd_vel /debug/approach_point /debug/ball_twist /debug/dsd/body_behavior /debug/dsd/hcm /debug/dsd/head_behavior /debug/dsd/localization /debug/used_ball /debug/which_ball_is_used /diagnostics /dynamic_kick/goal /DynamixelController/command /dynup/goal /field_boundary_in_image /field_boundary_relative /game_controller_connected /gamestate /goal_posts_in_image /goal_posts_relative /goal_relative /head_mode /imu/data_raw /imu_head/data /joint_states /line_mask_relative_pc /line_relative /line_relative_pc /move_base/BBPlanner/local_plan /move_base/NavfnROS/plan /move_base/current_goal /move_base/global_costmap/costmap /move_base/global_costmap/costmap_updates /move_base/goal /move_base/local_costmap/costmap /move_base/local_costmap/costmap_updates /move_base_simple/goal /obstacles_in_image /obstacles_relative /path_to_ball /pose_with_covariance /robot_state /rosout /server_time_clock /speak /strategy /system_workload /team_data /tf /tf_static'].
```

## Steps to Reproduce

1.
2.
3.
...

## Context (Environment)
Official HLVS setup, see this log: https://data.bit-bots.de/HLVS/2023/GAME_K-GD1-1_2023-03-17-07__itandroids_bitbots_99d633fc05/private_logs/ID_1/docker.log

## Possible Solution
- We could overwrite the existing output folder somehow
- We could somehow generate a unique output folder path, such that it can not already exist

Flova, 2023-03-26:
Maybe we can include a timestamp in the folder/file name. This also might be useful for organizational purposes.

jaagut, 2023-08-29:
Resolved with new record launch files